DCC-EX/EX-Installer #12: Bug Report: Clone/unknown devices don't display COM port

peteGSX, 2023-06-13:
### Version

The only one

### Bug description

When detecting attached devices, clones/unknown devices don't display the COM port they're attached to.

### Steps to reproduce the bug

Connect a clone device, progress to detect devices, and clone/unknown will be displayed without COM port.

### Expected behaviour

COM port should be displayed.

pmantoine, 2023-06-13:
[image]

Screenshot of two serial devices found, one of which (the bottom one) was a Mega+WiFI

pmantoine, 2023-06-13:
[image]
The devices in Device Manager

peteGSX, 2023-06-13:
Missed adding the COM port to the text variable for both unknown devices and devices that have multiple possible matches. Fixed, and will reflect in the next .exe build.

closed 2023-06-13 by peteGSX