DCC-EX/EX-Installer #19: Add options for setting TrackManager options

mstevetodd, 2023-06-14:
### Problem/idea statement

With the Devel version, the software supports several options for each track output: DCC, DC, DCX, Off, Booster
It would be great if this was exposed to the user through the install wizard.
If DC is selected, I'd also like to see a ROSTER() entry generated so the DC option shows clearly in throttle apps. Suggestion, when both are set to DC or DCX:
```
AUTOSTART SETLOCO(1) SET_TRACK(A, DC) DONE
ROSTER(1,"DC BLOCK A","/")
AUTOSTART SETLOCO(2) SET_TRACK(B, DC) DONE
ROSTER(2,"DC BLOCK B","/")
```

### Alternatives or workarounds

Add this code to myAutomation.h manually, requiring the user to be a programmer and handle file naming, programming syntax, etc. 
Many devs underestimate how difficult it is for non-devs to get case, quoting, commas, parenthesis, etc. exactly correct.

mstevetodd, 2023-06-18:
WIP screenshot of new options.
[image]

mstevetodd, 2023-06-20:
Pete suggested I add a switch to enable Track Manager settings. New screenshot:
[image]

Kcsmith0708, 2023-06-20:
Couple of thoughts,

When creating DC tracks we're starting a new type of technology and a way of thinking.
We have kicked around ideas of what to call these sections of track/blocks/districts.
and how to identify them from A -H.

We thought it would be best to try and use a term 'District' to describe the section of track vs Block.  Because we don't know how their current layout 'Block' is wired i.e. common ground or separate insulated tracks.
So we don't want them thinking in terms of Analogue Blocks.

Second, When creating ROSTER for DC Cabs  internal function keys don't operate and are labels only.
So for Analogue engines I create only one F0 key labeled "DC"  which shows on WiFi app throttles as DC.  This helps distinguish Analog engines on a throttle from DCC engines.
I also use this for DCC Motor & Lights Only Engines and use the first two labels as "Light"/DCC")  to distinguish it as a DCC motor only decoder but with a light function.  

In a mixed DC & DCC environment it becomes very easy and clear to know what engine type I have loaded in the throttle.

Regarding the Set Track mode on the EX-Installer.  Consider posting on the right side that it defaults to A MAIN and B PROG.

And if they'\ are settting them to DC in the Installer will it default to no address i.e. blank0 or will they individually set the District/Track address here also?

Or will it just default to Track A DC 1 and Track B DC 2. ?

mstevetodd, 2023-06-20:
@Kcsmith0708 I updated the suggestion and screenshots in the issue description to match what I'm doing now. 
Similar to you, I set a single function label to blank (since ED shows (currently) shows its defaults if you have no FKeys)
I also replaced BLOCK with TRACK.
TRACK A/B matches the output of the motor shield, I feel that DISTRICT gets us a bit too far into the "usage", which we can't control.

closed 2023-06-29 by peteGSX