DCC-EX/EX-Installer #71: [To Do]: Test build on Raspberry Pi

peteGSX, 2023-07-06:
### Task description

It'd be great if EX-Installer could run on Raspberry Pi.

@mstevetodd has tried to build on his image but it is failing to install pygit2 due to a libgit2 dependency issue.

Will try to see if I can replicate in Virtualbox to help figure out how to get past that to see if we can run it on RPi or not.

Start with this [download](https://www.raspberrypi.com/software/raspberry-pi-desktop/).

peteGSX, 2023-07-06:
Got to a point where it needs pygit2 1.6 or later, but this is not available in RPi Desktop. Need to look at building from source to see if it's worth the effort or not.

peteGSX, 2023-08-07:
This looks to be a no-go due to needing to build pygit2 from source most likely.

habazut, 2023-08-07:
Hm, on raspi you would have real git available, so really no need for pygit2

peteGSX, 2023-08-07:
True, but requires a reasonable amount of work specific to RPi rather than all the other platforms.

I'll reopen but it can go in the backlog for further consideration at some point in the future.

DavidNZ27, 2023-09-05:
Any news on this one Pete? I have a Raspberry Pi 4B running on the Raspberry PI OS 64bit (ARM aarch64 architecture however) that I just got for my layout that I am keen to run DCC-EX installer on. I followed the instructions here: https://github.com/DCC-EX/EX-Installer/blob/main/README.md#running-ex-installer with the virtual environment, and I got the EX-Installer to load, however when I went to Refresh Arduino CLI at the start, it errors out with the following error: "**2023-09-05 17:55:15.996 - ex_installer.arduino_cli - ERROR - Caught exception error: [Errno 8] Exec format error: '/home/jmri/ex-installer/arduino-cli/arduino-cli'**"  (I find the path in there odd as it is actually /home/jmri/EX_Installer/ex-installer/arduino-cli)

**(venv) jmri@jmripi:~/EX-Installer $ python -m ex_installer
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/jmri/EX-Installer/ex_installer/arduino_cli.py", line 126, in run
    if self.error:
AttributeError: 'ThreadedArduinoCLI' object has no attribute 'error'**

habazut, 2023-09-05:
There are packages used by the installer which for som unknown reason (demand?) have not been ported to the Raspi OS on ARM. For the time being the work load to support this on ARM is too much. You can use the arduino-cli instead. There is a wrapper called installer.sh which you can look at. Do `unset DISPLAY` then it will try to run in text mode without the graphical installer.

Regards,
Harald.

DavidNZ27, 2023-09-05:
Thanks for the response, that is what I was starting to wonder given the error message and understandable deeming the workload too much for the time being. It even looks like the Arduino CLI is getting left behind on Arm64 as the new 2.2 version doesn't have Arm64 builds, but V1.8 does, so have installed that and it works as I need, nowhere near as nice and seamless as EX-Installer, but it works, so I am happy!