DCC-EX/EX-Installer #100: [Feature Request]: Add ESP01/8266 AT firmware flashing

peteGSX, 2023-08-08:
### Problem/idea statement

With so many support issues around WiFi being reported in Discord, it would be beneficial to have esptool.py be able to be used via EX-Installer to flash the firmware on these chips, along with clear documentation on the connections and process.

### Alternatives or workarounds

Lots o Discord conversations with a very few specific individuals who know the process.

### Additional context

There are several different chip types that need to be catered for, with specific instructions for how to determine which type, and know what settings to apply.

peteGSX, 2023-08-10:
Need to add a "Tools" menu item, with a step through of the process with connections, then device identification, then flashing, and then a way to confirm/validate flashing was successful.

pmantoine, 2023-08-10:
Ideas:

1. Have users build standard CS stack
2. Have the user wire a digital output to GPIO0 so we can control "boot mode", or send codes to poke ESP8266 into boot mode?
3. Load serial passthrough sketch on CS so esptool.py on the PC can talk to the ESP8266 via CS
4. Detect FLASH ROM size
5. Erase totally ESP8266 flash
6. Flash 1.7.4 from our known good version with paramaters to suit the FLASH ROM size detected above

peteGSX, 2024-02-09:
No further traction on this, and no simple way to incorporate esptool.py into the installer either, and the workarounds with the Makerfabs shield supplying the correct firmware etc. seem to have solved the need.

closed 2024-02-09 by peteGSX