DCC-EX/EX-Installer #112: Bug Report: Installer does not update the Wi-Fi Name

sucharithv, 2023-10-24:
### Version

0.0.16

### Bug description

When a home WiFi SSID is provided it does not get written to the sketch. It is always going for the Access Point mode. I see that the network SSID shows as DCCEX_SAYS_BROKEN_FIRMWARE, and it is trying to set using the password provided for my WiFi.

I understand we need the v1.7.4 firmware for the MakerFab WiFi Shield to resolve other issues with WiFi bt it is not even attempting to connect to my home WiFi. If I do an advanced install using Arduino IDE and set the WiFI Name it successfully connects.

Log Form Monitor

<* License GPLv3 [fsf.org](http://fsf.org/) (c) [dcc-ex.com](http://dcc-ex.com/) *>
<* No I2C Devices found *>
<* MCP23017 I2C:0x20 Device not detected *>
<* MCP23017 I2C:0x21 Device not detected *>
<* Found PORTB pin 13 *>
<* Fault pin = 59 invert 0 *>
<* Pin 55 Max 4996mA (984) *>
<* Found PORTB pin 12 *>
<* Fault pin = 58 invert 0 *>
<* Pin 54 Max 4996mA (984) *>
<= A MAIN>
<= B PROG>
<* ++ Wifi Setup Try 1 ++ *>
<* Wifi Check: [+IPD\0] *>
<* TIMEOUT after 200ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
\r\nOK\r\n<* Found in 2ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
\r\nOK\r\n<* Found in 3ms *>
<* Wifi Check: [AT version:\0] *>
AT+GMR
AT version:<* Found in 4ms *>
2.2.0.0-<* Wifi Check: [\r\nOK\r\n\0] *>
dev(s-57dd9a4 - ESP8266 - Feb  9 2021 06:34:21)
SDK version:v3.4-rc
compile time(8e83329):Feb 26 2021 19:32:09
Bin version:2.0.0(WROOM-02-N)

OK
<* Found in 17ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CWJAP_CUR?\r\n\r\nERROR\r\n<* TIMEOUT after 2000ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CWMODE=1\r\n\r\nOK\r\n<* Found in 9ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CWMODE=2\r\n\r\nOK\r\n<* Found in 11ms *>
<* Wifi Check: [+CIFSR:APMAC,"\0] *>
AT+CIFSR
+CIFSR:APIP,"192.168.4.1"
+CIFSR:APMAC,"<* Found in 6ms *>
62:01:94:43:3b:d3<* Wifi Check: [\r\nOK\r\n\0] *>
"

OK
<* Found in 4ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CWSAP="**DCCEX_SAYS_BROKEN_FIRMWARE**","**myPassword**",1,4\r\n\r\nOK\r\n<* Found in 22ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CIPSERVER=0\r\n\r\nOK\r\n<* Found in 7ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CIPMUX=1\r\n\r\nOK\r\n<* Found in 4ms *>
<* Wifi Check: [\r\nOK\r\n\0] *>
AT+CIPSERVER=1,2560\r\n\r\nOK\r\n<* Found in 10ms *>
<* Wifi Check: [IP,"\0] *>
AT+CIFSR
+CIFSR:APIP,"<* Found in 4ms *>
192.168.4.1"<* LCD4:192.168.4.1 *>
<* Wifi Check: [\r\nOK\r\n\0] *>

+CIFSR:APMAC,"62:01:94:43:3b:d3"

OK

### Steps to reproduce the bug

1. Run the Installer
2. Select EX 8874_SHIELD
3. Enable have Wifi
4. Go to Wifi Options and select second option
5. Provide SSID and Password
6. Also tried Use Advanced Options which shows the myConfig.h files 
7. Compile and Load

### Expected behaviour

Expected to connect to my WiFi SSID Provided in Advanced Options

habazut, 2023-10-24:
DCC-EX does not try to use the firmware version 2.2.0 of the ESP8266 on the shield for real as it does not work. With 2.2.0 you can only connect your Wifi throttle once and all other connection attempts will fail. So you have to upadate the firmware to one of the ones that do work. We recommend you use firmware 1.7.4 on the Wifi shield as you correctly have observed. If you want to verify the bug, connect to the SSID "DCCEX_SAYS_BROKEN_FIRMWARE" with "myPassword" and see how it does only work **once** for  yourself.

sucharithv, 2023-10-24:
Are you saying that it will not connect to my home WiFi until I downgrade the firmware? 
I did go through this link and there are many sections missing https://dcc-ex.com/support/wifi-at-version.html#gsc.tab=0 
All the instructions I see need an USB TTL Adapter or a Uno board. 

Can you provide some help by pointing me to instructions on how to update Makerfab Wi-Fi Shield using just a mega 2560 board?

habazut, 2023-10-24:
2.2.0 does not work well. Neither as AP nor as STA.

You can wire an Uno or Mega (they are really identical in that respect) as described in section 3.2 here:
https://cordobo.com/2300-flash-esp8266-01-with-arduino-uno/
What they don't say there is that the purple wire to GPIO0 decides if it boots normal (GPIO0=3.3V) or in reflash mode (GPIO0=GND). Normal mode can be used to test if it reacts on AT+GMR to see the version. Thats at 115200 baud. The boot loader which says info about the hardware on the other hand  does give that info in 57600 baud.
If you are successful at 5. above, put purple wire to GND and jump to the software procedure below, sections 1b and 1d:
https://dcc-ex.com/reference/hardware/microcontrollers/wifi-mega.html#gsc.tab=0
Then if you think you have flashed successfully move the purple wire to 3.3V and check that you have the 1.7.4 version with AT+GMR.

closed 2024-02-09 by peteGSX