DCC-EX/EX-Installer #123: Bug Report:

higaski, 2024-01-23:
### Version

latest

### Bug description

The Linux version currently linked over on the [EX-Installer](https://dcc-ex.com/download/ex-commandstation.html#ex-installer) page gives me a `_tkinter.TclError`.

```sh
╰─λ ./EX-Installer-Linux64
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-sub-pixel-rgb.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-yes-antialias.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/48-spacing.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Traceback (most recent call last):
File "ex_installer/__main__.py", line 113, in <module>
File "ex_installer/__main__.py", line 64, in main
File "ex_installer/ex_installer.py", line 114, in __init__
File "tkinter/__init__.py", line 3272, in __init__
File "tkinter/__init__.py", line 2567, in __init__
_tkinter.TclError: failed to allocate font due to internal system font engine problem
[142245] Failed to execute script '__main__' due to unhandled exception!
```

Some sys info

```sh
╰─λ uname -a
Linux archlinux 6.7.0-zen3-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 13 Jan 2024 14:36:54 +0000 x86_64 GNU/Linux
```

### Steps to reproduce the bug

1. Download the latest version from the EX-Installer page
2. chmod +x EX-Installer-Linux64
3. ./EX-Installer-Linux64

### Expected behaviour

The installer should execute.

peteGSX, 2024-01-23:
What flavour of Linux is this? There's a known situation with Linux Mint where it will generate a bunch of warnings about fonts similar to what you see, but it still runs successfully and the warnings can be safely ignored.

From what I see in the error (and a bunch of searching I did for the Linux Mint issue) it's likely the system font configuration that's the issue here.

EX-Installer uses Helvetica which apparently may not always be installed on Linux distributions by default, and if the font system isn't configured to use a substitute, it may cause issues.

higaski, 2024-01-24:
It's KDE. I just checked and I've got "Adobe Helvetica" installed.

Also tried this:
https://stackoverflow.com/questions/45499865/trouble-allocating-fonts-in-a-tcl-tk-based-ubuntu-snap-app

Didn't do anything.

higaski, 2024-01-31:
Opened an issue over at the fontconfig repo as well:
https://gitlab.freedesktop.org/fontconfig/fontconfig/-/issues/404

peteGSX, 2024-02-02:
Same bug with Manjaros Linux as per https://github.com/DCC-EX/EX-Installer/issues/126.

Investigating options to prevent these exceptions if Helvetica is not installed/available.

As it turns out, Helvetica is not installed in Windows either, and it performs a seemless substitution, possibly with Arial.

Best option at the moment seems to be not to use Helvetica, and to use probably Arial on Windows, and FreeSans on Linux. Need to find the right option for Mac but possibly Arial as well.

habazut, 2024-02-02:
Could you install the following packages:
git
python3-tk
pythong3.8-venv
curl
(packages might be named slightly different)
and then run installer.sh and report if that starts a working installer or not.
Harald.

peteGSX, 2024-02-07:
According to the issue opened by @higaski over at fontconfig, the issue is the build is done on a system with an older version of fontconfig.

I've validated this is the case, as I've been building on Ubuntu 18 to try to ensure it runs on older Linux flavours, however this is obviously causing issues.

Have tested with Ubuntu 20 and the issue is resolved, so from the next release (which will be 0.0.17), these will be built on Ubuntu 20 which should resolve this issue.

higaski, 2024-02-08:
Thanks, works for me.

closed 2024-02-26 by peteGSX