DCC-EX/EX-Installer #143: [Feature Request]: Allow bypass of microcontroller selection

FrightRisk, 2024-03-06:
### Problem/idea statement

There are many times, like when experimenting, working offline, or doing support where you need to see what screens a user would see, and do not have a microcontroller handy. Need to me able to skip the micro selection to actually uses the features of the installer while offline

### Alternatives or workarounds

N/A

FrightRisk, 2024-05-14:
Maybe actually call this "offline mode" and even allow it to compile and save somewhere, but not upload. This could even have a "secret" hotkey if we just need it internally for testing

peteGSX, 2024-06-10:
Won't publicly document this, but running with -F or --fake will flag a USB device on either COM10 (Windows) or /dev/ttyUSB10 (Linux/macOS) if no devices are detected. This will allow running through the entire process including compilation, but will skip upload and flag success, enabling config backup to be performed as well.

closed 2024-06-10 by peteGSX