DCC-EX/EX-Installer #155: [To Do]: Improve Git process and upgrade order process

peteGSX, 2024-06-07:
### Task description

Diff:

`
diff --git a/ex_installer/arduino_cli.py b/ex_installer/arduino_cli.py
index afa7237..f0520c8 100644
--- a/ex_installer/arduino_cli.py
+++ b/ex_installer/arduino_cli.py
@@ -202,7 +202,7 @@ class ArduinoCLI:
     # Dictionary for additional board/platform support for the Arduino CLI
     extra_platforms = {
         "Espressif ESP32": {
-            "platform_id": "esp32:esp32",
+            "platform_id": "esp32:esp32@2.0.17",
             "url": "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"
         },
         "STMicroelectronics Nucleo/STM32": {
diff --git a/ex_installer/manage_arduino_cli.py b/ex_installer/manage_arduino_cli.py
index 98e3fbc..4dda9e5 100644
--- a/ex_installer/manage_arduino_cli.py
+++ b/ex_installer/manage_arduino_cli.py
@@ -269,7 +269,14 @@ class ManageArduinoCLI(WindowLayout):
             elif self.process_status == "error":
                 self.process_error(self.process_topic)
                 self.restore_input_states()
-        elif self.process_phase == "update_index" or self.process_phase == "install_packages":
+        elif self.process_phase == "update_index":
+            if self.process_status == "success":
+                self.process_start("upgrade_platforms", "Upgrading Arduino platforms", "Manage_CLI")
+                self.acli.upgrade_platforms(self.acli.cli_file_path(), self.queue)
+            elif self.process_status == "error":
+                self.process_error(self.process_topic)
+                self.restore_input_states()
+        elif self.process_phase == "upgrade_platforms" or self.process_phase == "install_packages":
             if self.process_status == "success":
                 if self.package_dict:
                     package = next(iter(self.package_dict))
@@ -290,15 +297,9 @@ class ManageArduinoCLI(WindowLayout):
                     self.process_start("install_libraries", "Install Arduino library " + library, "Manage_CLI")
                     self.acli.install_library(self.acli.cli_file_path(), library, self.queue)
                 else:
-                    self.process_start("upgrade_platforms", "Upgrading Arduino platforms", "Manage_CLI")
-                    self.acli.upgrade_platforms(self.acli.cli_file_path(), self.queue)
-            elif self.process_status == "error":
-                self.process_error(self.process_topic)
-                self.restore_input_states()
-        elif self.process_phase == "upgrade_platforms":
-            if self.process_status == "success":
-                self.process_start("refresh_list", "Refreshing Arduino CLI board list", "Manage_CLI")
-                self.acli.list_boards(self.acli.cli_file_path(), self.queue)
+                    self.process_stop()
+                    self.process_start("refresh_list", "Refreshing Arduino CLI board list", "Manage_CLI")
+                    self.acli.list_boards(self.acli.cli_file_path(), self.queue)
             elif self.process_status == "error":
                 self.process_error(self.process_topic)
                 self.restore_input_states()
`

habazut, 2024-06-07:
Fix.patch in attachment.

habazut, 2024-06-07:
try again

peteGSX, 2024-06-07:
`
diff --git a/ex_installer/git_client.py b/ex_installer/git_client.py
index 85bac66..674aecb 100644
--- a/ex_installer/git_client.py
+++ b/ex_installer/git_client.py
@@ -240,8 +240,7 @@ class GitClient:
         """
         Gets the ref for the named branch
         """
-        branch = repo.lookup_branch(name)
-        refname = repo.lookup_reference(branch.name)
+        refname = repo.resolve_refish(name)
         GitClient.log.debug("Branch %s ref is %s", name, refname)
         return refname
 
`

closed 2024-06-09 by peteGSX