DCC-EX/EX-Installer #184: ESP32 "incorrect password"

mvackel, 2024-09-23:
### Version

0.0.20

### Bug description

Using ESP32 Dev Kit the installation of Ex-CommandStation doesn't allow wifi authentication with message "incorrect password". On Android, the message is "Authentication problem". I've tried firmware versions 5.07, 5.09, 5.77 and 5.78. Also tried Ex-installer versions 0.0.19 and on 3 different computers.
When installing Ex-CommandStation from VSCode, from the sources (v 5.07) everything works perfectly. 

### Steps to reproduce the bug

All default installation. Select ESP32 Dev Kit and the "STANDARD_MOTOR_SHIELD". Every option on the default.

### Expected behaviour

The installer completes with a "Success" message, but when we try to connect to the wifi, there is a "incorrect password" message.

habazut, 2024-09-24:
Do you have any 'not so usual' chars in your Wifi password?
Does the password look correct in the config.h generated by the installer? If not, what did the installer get wrong?

Harald.

mvackel, 2024-09-24:
The password is the default one (PASS_xxxxxx where xxxxxx are the last 6 digits from the MAC address). This was confirmed on the serial monitor. The installer doesn't report anything wrong at all. I suspect the cause may be the Espressif framework version, as they are different in the installer (2.0.17) and VSCode (2.0.9).

habazut, 2024-09-24:
We definitely do use the 2.0.17 with success by default. As well in the installer as in all my manual builds.
Typical startup log from an ESP32 built with ESP package 2.0.17:
```
20:00:33.867 -> [   206][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
20:00:33.867 -> [   281][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
20:00:33.867 -> [   301][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
20:00:33.867 -> [   302][V][WiFiGeneric.cpp:344] _arduino_event_cb(): STA Stopped
20:00:33.867 -> [   315][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 3 - STA_STOP
20:00:33.867 -> [   322][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
20:00:33.867 -> <* Wifi[   336][V][WiFiGeneric.cpp:393] _arduino _event_cb(): AP Started
20:00:33.867 -> in AP mode *>
20:00:33.867 -> <@ 0 5 "Wifi: DCCEX_76e614">
20:00:33.867 -> <@ 0 6 "PASS: PASS_76e614">
20:00:33.867 -> <@ 0 7 "IP: 192.168.4.1">
20:00:33.867 -> [   351][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
20:00:33.903 -> [   370][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=2560, ...)
20:00:33.903 -> <* Server will be started on port 2560 *>
```
where the lines with [ come from the ESP32 core debug level = verbose.
So I wonder what the problem could be you are seeing.

Harald.

pmantoine, 2024-09-25:
Later development versions of DCC-EX stipulate the framework version in the VS Code platformio.ini file, thusly:

[env:ESP32]
; Lock version to 6.7.0 as that is
; Arduino v2.0.16 (based on IDF v4.4.7)
; which is the latest version based
; on IDF v4. We can not use IDF v5.
platform = espressif32 @ 6.7.0

... which uses v2.0.16 of the Espressif core. Feel free to add this to the v5.0.7 version of platformio.ini to see it work.

I also just changed this to 6.8.1 which is the spec for v2.0.17 of the Espressif core, and that too connected just fine.

platform = espressif32 @ 6.8.1

Can you please show us the startup log for the ESP32 when you have v5.0.7 loaded with EX-Installer?

mvackel, 2024-09-26:
I've just tried again, with nothing changed, and everything works now. This is very strange, as I'm struggling with this in the last few days, with 3 different boards, 2 different computers. The only thing I can think now that could have caused this is the wi-fi network. Dcc-ex uses a fixed channel (channel 1) for the ESP32 and, maybe, this channel was somehow cluttered... but how does it worked with version 5.07 uploaded by VSCode? I can't see any reasonable explication. Can you leave this thread open for 2 or 3 more days, so I can make more tests?

pmantoine, 2024-09-27:
It is odd you exprienced this... the ESP32 wifi support has been stable a long time now (even prior to v5.0.x release), despite the ESP32 being in beta test.

I have assumed you're just using the regular Espressif ESP32-WROOM Dev Kit?

No problem at all leaving this open for you while you investigate further.

Paul