DCC-EX/EX-Installer #189: [Support Request]:

JBERRY812, 2025-01-27:
### Version

Elegoo Mega 2560 Rev3

### Issue description

My ommand station on my DCC EX system connected to JMRI on decoder Pro with Wi throttle as a throttle works fine to power up and use my locos on the main track.  Howver, I cannot for the life of me get the system to program my locomotives.  I have them on the practice track plug in on the motherboard, and with the programming track set on the on the Decoder Pro.  But I have tried at least 15 times, and I get the same 308 response.  What should I do?

Ash-4, 2025-01-27:
If you are having 308 errors, use a USB connection to the computer running DecoderPro.
Then you can utilize the JMRI DCC++ Send Command window and the DCC++ Traffic Monitor (use the checkbox 'Show raw data' and deselect 'Translated')

Send the commands
<D ACK ON>
<R 8>

https://dcc-ex.com/reference/tools/diagnostic-d-ack-command.html#more-detail
Attach a log with the results.  Also advise on what uC and motor shield are used for the DCC-EX command station.

The discord server is preferred for assistance.
https://dcc-ex.com/support/index.html#preferred-option-discord

Asbelos, 2025-01-28:
Please send the command <D ACK ON> before <R 8> 
This turns on verbose logging of the cv read process and from that log we can usually tell what the problem is.

Ash-4, 2025-01-28:
There was/is a problem in displaying the command to turn on the diagnostics - at least as shown in my browser.
`<D ACK ON>
<R 8>`

In this reply, I used the button -- apparently it added the grave accent before/after.