DCC-EX/EX-Installer #192: [Support Request]: Could not open COM5, the port doesn't exist

ClayAAllison, 2025-04-16:
### Version

CommandStation-EX.ino.bin"  esptool.py v4.5.1

### Issue description

A fatal error occurred: Could not open COM5, the port doesn't exist
I have Windows 11 and no pending updates.  I have clicked off the "Allow computer to turn off port" check box on all available USB ports.
I removed the software and reinstalled it and scanned my machine for any malware.  I have rebooted after each try.  
I installed a USB driver CH341SER

I am stumped. I just got the SB1 yesterday.  It has DCC-EX v5.4.6  serial no ce7118.  

Any assistance would be appreciated.  
Thank you in advance.
Clay Allison

Asbelos, 2025-04-17:
Windows does not always use the same COM port number every time. You may find it on a different port.

You can only run one program talking to the command station port. Please make sure you close any other program that may be using the port, such as JMRI, EX-Webthrottle or tha Arduino IDE.

ClayAAllison, 2025-04-17:
Thank you very much for your response.  I actually found the solution on the website.  I said I should download the USB driver and KaPoofKa it worked just fine.

Again, thank you very much for getting back to me,

Sincerely,
Clay Allison a very satisfied customer
Houston Texas

closed 2025-05-10 by peteGSX