DCC-EX/EX-Installer #193: [Support Request]: Failed to connect to Espressif device: No serial data received.

gslance, 2025-05-07:
### Version

esptool.py version 3.3.2 (and also 4.4)

### Issue description

Fresh out the box I was able to flash the device but it did not properly turn on after its automated reset. Display remained off, no access point or serial. After this, I was not able to re-flash the device anymore using EX Installer. 

Every future attempt to install will either result in the device not being recognized (descriptor issue, windows warning) or most recently the message `A fatal error occurred: Failed to connect to Espressif device: No serial data received.` Sometimes it will not even recognize the COM port, even though it does appear in the Device Manager.

- I've already installed the USB driver (CH341)
- No windows updates, multiple rests (I'm on windows 11)
- Device turns on, green 3V3 light turns on when connecting through USB
- Tried multiple different cables

I'm at a loss, this thing came in today.

Ash-4, 2025-05-07:
The problem reported makes me wonder if EX-Installer was downloaded and run from the link on the DCC-EX website.
https://dcc-ex.com/download/ex-commandstation.html#ex-installer

The Windows x64 version should not result in questions about which version of esptool.py you are using.

Try the discord server for an interactive discussion.
https://dcc-ex.com/support/index.html#preferred-option-discord

gslance, 2025-05-07:
Yes, I used the Auto Installer from the website to get the x64 version appropriate for my Windows installation. Like I said, the first time around the installer seemingly succeeded but my device never turned on properly after its reboot.

Afterwards, problems kept arising from the installer attempting to re-flash it. My plan was to use the development version instead of the Recommended. I've then tried different esptool.py versions to verify if the problem has some consistency and figure out whether it's software or hardware. That's why I bothered to mention the esptool versions I checked out.

pmantoine, 2025-06-12:
Was this resolved? If not, let me know how I can help.