DCC-EX/EX-Installer #209: [Support Request]: Installer wont load the latest dccex program.  Says installer id deleted

stonedragonbonsai, 2025-09-01:
### Version

Latest DCC EX installer

### Issue description

When I try to load product I get an error message saying theEX-commandstation-installer.exe is deleted and do I want to override.
The log then shows the following.
2025-09-01 17:39:14.138 - ex_installer.git_client - ERROR - Local file changes in pygit2.Repository('C:/Users/User/ex-installer/CommandStation-EX/.git/')
2025-09-01 17:39:14.138 - ex_installer.git_client - ERROR - ['EX-CommandStation-installer.exe (Deleted)']
2025-09-01 17:39:14.138 - ex_installer.select_version_config - ERROR - Local repository file changes: ['EX-CommandStation-installer.exe (Deleted)']
2025-09-01 17:39:25.181 - ex_installer.ex_installer - CRITICAL - Uncaught exception: Traceback (most recent call last):
  File "tkinter\__init__.py", line 1921, in __call__
  File "customtkinter\windows\widgets\ctk_button.py", line 554, in _clicked
    self._command()
  File "ex_installer\select_product.py", line 97, in <lambda>
  File "ex_installer\select_product.py", line 143, in check_product_device
  File "ex_installer\ex_installer.py", line 252, in switch_view
  File "ex_installer\select_version_config.py", line 103, in set_product
  File "ex_installer\select_version_config.py", line 195, in setup_local_repo
  File "ex_installer\select_version_config.py", line 430, in resolve_local_changes
  File "ex_installer\git_client.py", line 341, in git_hard_reset
AttributeError: '_pygit2.Commit' object has no attribute 'oid'

The program then crashes and I exit.
 
It was working until I upgraded to win 11.

peteGSX, 2025-09-01:
Do you have an antivirus program installed? I suspect it's falsely detecting that .exe file as a trojan and quarantining it. You may need to get your antivirus software to ignore it. At some point that file will be deprecated and removed as it's no longer required, but at the moment it will cause the issues you're seeing if antivirus falsely detects it as a trojan.

stonedragonbonsai, 2025-09-01:
> No changes are permitted in the folders belonging to EX-Installer.
> Delete the folder and it will be restored when installer is run.
> C:\Users_____\ex-installer\

Deleting the folder and rerunning the installer worked.  Thank you very much.
I haven't played with the program since last using it successfully.  The only changes I have made was to upgrade to windows 11.
Very strange.

closed 2025-09-02 by peteGSX